snapcrafters/arduino #1: Themes!

mbauer599, 2020-03-01:
By default, only the default theme is availiable. I was also unable to add aditional themes. 

Either documentation around this or included themes support would be ideal. I like dark mode.

galgalesh, 2020-03-01:
Hm, I didn't know Arduino had themes. Could you point me to docs explaining how to install them?

mbauer599, 2020-03-01:
This is really what i'm after:
https://create.arduino.cc/projecthub/rahulkhanna/dark-theme-for-arduino-ide-17c001

I've only ever replaced the default theme to get this, but there is actually a theme drop down in preferences under settings. 

The issue with following that guide is the read only file systems. If themes can exist side by side then you could conceivably include both themes and allow for the selection of such, or have the themes pulled from user space so they can build/deploy their own. Either way would be a nice add.

galgalesh, 2020-03-01:
Ok, I'll add this theme to the default install. Any other themes you think I should add?

PS: You can add the theme yourself to your own install by zipping the _contents_ of the `theme` folder and putting the zip in `~/snap/arduino/current/Arduino/theme` (you might need to create this diredtory). Make sure to zip the contents, not the directory itself, so the theme files are in the root of the zip instead of in the `theme` folder.

mbauer599, 2020-03-01:
Good to know! This theme is really the only one I have ever used so if it's there by default I know a lot of people who will be happy with that, but i'm sure there are some other good ones out there. I would add the comment above about adding the theme yourself to the Readme, just so everyone knows how. If you don't get to that before me then I can PR with instructions, but it could be a while before I get to it.

mbauer599, 2020-03-01:
Testing it out on edge and everything here looks good. Thank you for the quick response on this one.

galgalesh, 2020-03-01:
Great thanks!

Fixed in https://github.com/snapcrafters/arduino/commit/1c78ec2a9e029b45d260d5137b5d166113add351, closing this issue.

closed 2020-03-01 by galgalesh