snapcrafters/arduino #4: java.nio.file.AccessDeniedException when installing libraries

trevorpeacock, 2020-04-28:
When I try to install a library through "Library Manager" (Tools->Manage Libraries) I get the following error
```
 /home/user/snap/arduino/7/Arduino/libraries/arduino_173950
java.lang.RuntimeException: java.nio.file.AccessDeniedException: /home/user/snap/arduino/7/Arduino/libraries/arduino_173950
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$4(LibraryManagerUI.java:248)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.nio.file.AccessDeniedException: /home/user/snap/arduino/7/Arduino/libraries/arduino_173950
	at sun.nio.fs.UnixException.translateToIOException(UnixException.java:84)
	at sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:102)
	at sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:107)
	at sun.nio.fs.UnixFileSystemProvider.createDirectory(UnixFileSystemProvider.java:384)
	at java.nio.file.Files.createDirectory(Files.java:674)
	at java.nio.file.Files.createAndCheckIsDirectory(Files.java:781)
	at java.nio.file.Files.createDirectories(Files.java:767)
	at processing.app.helpers.FileUtils.createTempFolder(FileUtils.java:118)
	at processing.app.helpers.FileUtils.createTempFolder(FileUtils.java:106)
	at processing.app.helpers.FileUtils.createTempFolder(FileUtils.java:102)
	at cc.arduino.contributions.libraries.LibraryInstaller.performInstall(LibraryInstaller.java:169)
	at cc.arduino.contributions.libraries.LibraryInstaller.install(LibraryInstaller.java:125)
	at cc.arduino.contributions.libraries.LibraryInstaller.install(LibraryInstaller.java:117)
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$4(LibraryManagerUI.java:239)
	... 1 more
```

galgalesh, 2020-04-28:
Thanks for the report!

I am able to install any library I tried. Can you give some more information to help me reproduce the problem?

* Which library are you trying to install?
* Where do you see that error?
* Which OS and version are you using?

galgalesh, 2020-04-28:
I also see you're running a very old version of the snap. You're running revision 7 while the current stable is revision 14. Could you try to refresh the snap (`snap refresh arduino --stable`) and try again?

trevorpeacock, 2020-04-29:
Thanks so much @galgalesh . I got it running thanks to you pointing me in the right direction. Sorry for not providing more detail, I should know better. I also didn't expect you to troubleshoot my problem, I was hoping I'd be able to help you help someone else, I thought I was already running the latest version. I wouldn't have created an issue otherwise.

### Backgound
Running a freshly installed Ubuntu 20.04
```
$ snap version
snap    2.44.3+20.04
snapd   2.44.3+20.04
series  16
ubuntu  20.04
kernel  5.4.0-26-generic
$ snap list
Name                             Version                     Rev   Tracking         Publisher          Notes
arduino                          1.8.12                      14    latest/stable    merlijn-sebrechts  -
```

The output reported above appeared in the arduino gui log window (at the bottom of the screen, where build/programming logs are presented)
I was trying to install the `OneWire` library (Jim Studt, Tom Pollard, Robin James et. at.) and `DallasTemperature` (Miles Burton, Tim Newsome et. al.), however it tried a few others at random with the same result.

### Investigation
I had previously checked the ubuntu software gui to see if there were updates.. couldn’t find any mention of being able to update.
The command you suggested resulted in `snap "arduino" has no updates available`

Looking at `~/snap/arduino/` there as a `7` directory and a `14` directory, however when starting arduino from the ubuntu menu, it was still running from the 7 directory.

Running `snap remove arduino && snap install arduino` resolved the problem. I now only have the `14` directory, and ubuntu now starts the version from that directory.

I don't know if theres something funny happening with newer revisions not being activated or installed properly (specifically to do with registering the app in the ubuntu unity menu) when there's is a previous version installed, snaps are fairly foreign to me, I can't figure out how to install the previous revision to re-create the situation.
I'm happy for you to close this unless you'd like to dig into that problem, I'm happy to do some testing if you can walk me through it.

galgalesh, 2020-05-05:
@trevorpeacock Thanks again for your report and extensive description. I located the issue in the snap. When arduino generates and saves its config file, it adds a link to the snap data directory in the `sketchbook.path` variable. Instead of using the `current` symlink in the path, arduino pointed to the path of that specific revision (`snap/arduino/7` instead of `snap/arduino/current`). So new versions of the snap would still use the older data directory.

This is fixed in https://github.com/snapcrafters/arduino/commit/aae1a0347c18926d9ae74c65c6c982f0591cf79a

galgalesh, 2020-05-06:
This fix has been released to stable. Please let me know if you have any issues installing libraries in the future.

closed 2020-05-06 by galgalesh