snapcrafters/arduino #5: Unable to use python based tools with this snap

guneemwelloeux, 2020-05-12:
# Description
I tried to compile an ESP32 sketch with the snap version of Arduino IDE, and got an error saying `python` wasn't in my PATH. But it is there.
I think this is due to snaps running in a sandbox environment.

# Steps to reproduce
1. Install Arduino IDE from Ubuntu software
1. Add this URL to the board manager additional URL list in Preferences
1. Go to Tools, Board, Board manager, and look for esp32, and install it
1. Select an ESP32 board (I used MH ET Live ESP32DevKIT)
1. Open an example sketch for ESP32 (I used ESP32/AnalogOUT/LEDCSoftwareFade)
1. Click verify, and the error appears in the console at the end.

# Additional info
The esptool.py utility that is running at the end is used to flash the firmware onto the board, and uses python, but also requires the `serial` python package.

galgalesh, 2020-05-12:
Thanks for reporting this issue!

What is the URL that I should add to URL list?

galgalesh, 2020-05-12:
This is fixed in https://github.com/snapcrafters/arduino/commit/35ef2495ffdeb33273aa0575b9ed5ff608276e3c

I added python and the `pyserial` package, since esp32 requires that. If you need any other Python 2 dependencies, you can install them using `arduino.pip install <packagename>`. These packages will be installed in the sandbox of the Arduino snap so you don't have to worry about these dependencies (and the now unsupported Python 2 itself) messing up your host system.

@guneemwelloeux could you test this with the snap in the edge channel? `snap install arduino --edge` or `snap refresh arduino --edge` if it's already installed. If you have both the snap and the deb installed, you can run `snap run arduino` to make sure you're testing the snap and not the .deb package.

guneemwelloeux, 2020-05-13:
Hi @galgalesh 

Thank you for the quick turnaround. I tested the edge channel version of the snap, and can happily report that the issue is fixed.

For information, I had downloaded the Arduino IDE from the website, and installed it. It was working with Python3 after I installed the `python3-serial` and `python-is-python3` packages (running Ubu 20.4). So it appears that the espressif python tools seem to be working with Python 3, and maybe this fact could be used in this snap so that it embeds python3 instead of python2...

galgalesh, 2020-05-13:
That's good to know, thanks! I switched to python 3. Could you test again with the snap in the `edge` channel?

galgalesh, 2020-05-13:
PS: I'm not sure if you're using OTA upgrades, but that might also require more permissions for the sandbox. If you have any issues with OTA, let me know. I don't have a compatible board myself so I can't really test it out.

guneemwelloeux, 2020-05-15:
I'm not sure what you mean with OTA in this context (does it refer to the comment about being in the dialout group?), but I tested the latest version of the snap, with python3, and it works very well. Thanks

galgalesh, 2020-05-15:
This is what I mean with OTA upgrades: https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html

Given that this is over wifi; not serial, the snap might require additional permissions depending on how it works exactly. I don't have a board which has wifi, so I can't test it myself, but I've seen a review mention that it doesn't work.

[image]

galgalesh, 2020-05-15:
The fix is now released to stable. Thanks a lot for your help!

closed 2020-05-15 by galgalesh