snapcrafters/arduino #6: Unable to upload sketch Over The Air (ESP boards)

guneemwelloeux, 2020-05-16:
Following the conversation about OTA sketch upload in a previous issue, I can confirm that the snap is not allowing any upload to a network port.
In the case of the ESP32 and the ESP8266, the upload fails with the following error:
```
09:21:45 [ERROR]: Listen Failed
```
After having a look at the script doing the upload, and modifying the `except` block to show the undelying error, I can see that the error is caused by a missing permission when the script is calling `socket.bind`.

Based on this, I reckon the snap in missing the `network-bind` plug.

galgalesh, 2020-05-16:
The latest version in stable has the `network-bind` plug. Can you try again?

guneemwelloeux, 2020-05-16:
OTA updates work perfectly with the latest stable version!
Thanks!

closed 2020-05-16 by galgalesh