snapcrafters/arduino #7: Esp8266 "the selected serial port _ does not exist or your board is not connected"

Razzo1987, 2020-05-16:
Hi,
I'm using a NodeMCU 1.0 on /dev/ttyUSB0 and when I try to Upload sketch I receive this error:
```
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "/home/razzo/snap/arduino/29/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "/home/razzo/snap/arduino/29/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/esptool/esptool.py", line 2890, in main
    esp.connect(args.before)
  File "/home/razzo/snap/arduino/29/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/esptool/esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
_
the selected serial port _
 does not exist or your board is not connected
```
I can't find a solution

galgalesh, 2020-05-16:
Can you run the following commands one by one and show me the output?

```
lsusb
snap connections arduino
snap version
ls /dev/tty*
```

Razzo1987, 2020-05-16:
Yes:
```
$ lsusb
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 04f2:b3fd Chicony Electronics Co., Ltd HD WebCam (Asus N-series)
Bus 003 Device 004: ID 13d3:3402 IMC Networks Bluetooth USB Host Controller
Bus 003 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 003 Device 002: ID 0b0e:0313 GN Netcom Jabra EVOLVE 30 II
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
```
$ snap connections arduino
Interface               Plug                     Slot                              Notes
content[gtk-2-engines]  arduino:gtk-2-engines    gtk2-common-themes:gtk-2-engines  -
content[gtk-2-themes]   arduino:gtk-2-themes     gtk-common-themes:gtk-2-themes    -
content[icon-themes]    arduino:icon-themes      gtk-common-themes:icon-themes     -
content[sound-themes]   arduino:sound-themes     gtk-common-themes:sound-themes    -
home                    arduino:home             :home                             -
network                 arduino:network          :network                          -
network-bind            arduino:network-bind     :network-bind                     -
raw-usb                 arduino:raw-usb          :raw-usb                          -
removable-media         arduino:removable-media  -                                 -
serial-port             arduino:serial-port      -                                 -
unity7                  arduino:unity7           :unity7                           -
x11                     arduino:x11              :x11                              -
```
```
$ snap version
snap    2.44.3+20.04
snapd   2.44.3+20.04
series  16
ubuntu  20.04
kernel  5.4.0-29-generic
```
```
$ ls /dev/tty*
/dev/tty    /dev/tty14  /dev/tty20  /dev/tty27  /dev/tty33  /dev/tty4   /dev/tty46  /dev/tty52  /dev/tty59  /dev/tty8       /dev/ttyS12  /dev/ttyS19  /dev/ttyS25  /dev/ttyS31  /dev/ttyUSB0
/dev/tty0   /dev/tty15  /dev/tty21  /dev/tty28  /dev/tty34  /dev/tty40  /dev/tty47  /dev/tty53  /dev/tty6   /dev/tty9       /dev/ttyS13  /dev/ttyS2   /dev/ttyS26  /dev/ttyS4
/dev/tty1   /dev/tty16  /dev/tty22  /dev/tty29  /dev/tty35  /dev/tty41  /dev/tty48  /dev/tty54  /dev/tty60  /dev/ttyprintk  /dev/ttyS14  /dev/ttyS20  /dev/ttyS27  /dev/ttyS5
/dev/tty10  /dev/tty17  /dev/tty23  /dev/tty3   /dev/tty36  /dev/tty42  /dev/tty49  /dev/tty55  /dev/tty61  /dev/ttyS0      /dev/ttyS15  /dev/ttyS21  /dev/ttyS28  /dev/ttyS6
/dev/tty11  /dev/tty18  /dev/tty24  /dev/tty30  /dev/tty37  /dev/tty43  /dev/tty5   /dev/tty56  /dev/tty62  /dev/ttyS1      /dev/ttyS16  /dev/ttyS22  /dev/ttyS29  /dev/ttyS7
/dev/tty12  /dev/tty19  /dev/tty25  /dev/tty31  /dev/tty38  /dev/tty44  /dev/tty50  /dev/tty57  /dev/tty63  /dev/ttyS10     /dev/ttyS17  /dev/ttyS23  /dev/ttyS3   /dev/ttyS8
/dev/tty13  /dev/tty2   /dev/tty26  /dev/tty32  /dev/tty39  /dev/tty45  /dev/tty51  /dev/tty58  /dev/tty7   /dev/ttyS11     /dev/ttyS18  /dev/ttyS24  /dev/ttyS30  /dev/ttyS9

```

galgalesh, 2020-05-16:
Hm, everything looks in order.

Have you tried holding the BOOT/FLASH button on the device while uploading? This article explains more: https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/

Razzo1987, 2020-05-17:
Thanks @galgalesh  I tried your solution but not solve.
Anyway I fond the problem: I'm using 2 DHT11 sensor, one have the data port connected to D8. I change the port to D6 and now work correctly.

SOLVED for me

galgalesh, 2020-05-17:
Awesome!

closed 2020-05-17 by galgalesh

DougLeary, 2020-12-10:
Changing the pin choice for the sensor definitely did not solve this problem. The problem just went away for some unknown reason.

Normanras, 2021-02-23:
Thank you for saying that, @DougLeary

Domin8668, 2021-11-14:
> Hm, everything looks in order.
> 
> Have you tried holding the BOOT/FLASH button on the device while uploading? This article explains more: https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/

Thanks for that link, helped me with a similar issue :)

LV-55, 2021-11-29:
> Changing the pin choice for the sensor definitely did not solve this problem. The problem just went away for some unknown reason.

OK, well here's a thing. I was following https://microcontrollerslab.com/esp8266-nodemcu-stepper-motor-websocket-web-server/ but had the above problem. Changing USB port and ESP8266 didn't change things. I could even get Blink to load. UNTIL I removed the connection from IN1 to D8, after which the code from the web-page loads, Blink loads ... Go figure.

SyedAhmedAlishah, 2021-12-30:
> OK, well here's a thing. I was following https://microcontrollerslab.com/esp8266-nodemcu-stepper-motor-websocket-web-server/ but had the above problem. Changing USB port and ESP8266 didn't change [...]

I had a similar problem that occurs when during code uploading if pin 8  is connected to something. It is similar to when you don't connect anything to 'TX and RX'  of Arduino during uploading otherwise it throws a similar error.