snapcrafters/arduino #8: Odd Menu Behaviour

pelfking, 2020-07-15:
I'm using your IDE version 1.8.13 on Ubuntu 20.04 LTS. This is an upgrade from my Ubuntu 18 setup that was working fine with a manually installed IDE.
I'm unable to access the extreme top and bottom of the fly out menus - for example the 'Boards Manager..' option and the down arrow on the 'Examples' menu. Because of this I'm unable to select add on boards or see examples from installed libraries.

I have tried using version 1.8.12 with the same issue being present. 
I have tried uninstalling your version and installing the one from arduino.cc - same issue.
I've tried making Java 8 available on my system, but I think that the IDE may supply it's own Java - it made no difference in any event, even though I think that the menuing system could be Java Swing.
I am at a loss to explain what's happening here.
[image]

galgalesh, 2020-07-15:
I can scroll through these menu's by hovering over the black arrows and by using the scroll wheel. Does this work for you?
 
[image]

pelfking, 2020-07-15:
Yes, I can scroll down using the 'wheel' (the edge of a trackpad in my case) but can no longer select the 'down arrow' to scroll down.

galgalesh, 2020-07-15:
That's really strange. For me, the "down arrow" gets selected when I move my mouse over it, and the list scrolls by moving my mouse there.

Since you're also having this issue with the version from arduino.cc, it doesn't seem to be an issue with the snap itself. This bugtracker is specific to the snap. Can you report the issue upstream and link the new issue here? I'll close the issue here then. Thanks!

https://github.com/arduino/Arduino/issues

closed 2020-07-16 by pelfking