snapcrafters/arduino #9: ESP32 CAM

arthurross88, 2020-07-22:
I just can't figure out what I'm doing wrong. Can someone help me? I get this output:
esptool.py v2.6
Traceback (most recent call last):
Serial port /dev/ttyUSB2
  File "/home/arthur/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 2959, in <module>
Connecting...
An error occurred while uploading the sketch
    _main()
  File "/home/arthur/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 2952, in _main
    main()
  File "/home/arthur/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 2653, in main
    esp.connect(args.before)
  File "/home/arthur/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 460, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False)
  File "/home/arthur/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 418, in _connect_attempt
    self._setDTR(False)  # IO0=HIGH
  File "/home/arthur/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 384, in _setDTR
    self._port.setDTR(state)
  File "/usr/local/lib/python2.7/dist-packages/serial/serialutil.py", line 599, in setDTR
    self.dtr = value
  File "/usr/local/lib/python2.7/dist-packages/serial/serialutil.py", line 469, in dtr
    self._update_dtr_state()
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 636, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIC, TIOCM_DTR_str)
IOError: [Errno 22] Invalid argument

galgalesh, 2020-07-22:
This bugtrackers is specifically for the snap package. It seems like you are using a different installation method. You could try if the snap solves the issue for you. You can install it using by running `snap install arduino` in the commandline and try it out using `snap run arduino` in the commandline. It also has a shortcut in the applications menu, but if you keep your current installation, you'll have two icons and it's not clear which one will be the snap).

Otherwise, the Arduino folks will be able to help you further:

* Either ask a question on their forum (https://forum.arduino.cc)
* Or create a bug on their bugtracker (https://github.com/arduino/arduino/issues)

Thanks!

galgalesh, 2020-07-26:
Closing now since it's an issue with the upstream installer. Please reopen if you have questions about the snap.

closed 2020-10-25 by galgalesh