snapcrafters/arduino #10: cannot open device: Permission denied

mmartinortiz, 2020-10-24:
Hi,

Before opening this issue, I've read the "Readme" in the Snap Store and search in the Internet to see how to fix it. I'm running:

- Ubuntu 20.04
- Arduino IDE 1.8.13

- [x] Installed the Arduino IDE via the Ubuntu Software as Snap. I've checked the permissions of the Snap, all are enabled. `Access serial port hardware` drop down list only shows `(disconnected)`
- [x] Added my user to the `dialout` group. 
- [x] Computer rebooted (several times). When I run the command `groups` I get `myuser adm dialout cdrom sudo dip plugdev lpadmin lxd sambashare docker`
- [x] Followed the steps to install my Arduino board (Arduino Uno Wifi rev2). Instructions [detailed here](https://forum.arduino.cc/index.php?topic=600558.msg4078155#msg4078155)

At the end, I reached to [this comment](https://forum.arduino.cc/index.php?topic=600558.msg4158135#msg4158135). I've checked the `udev` rules created by the Snap, I see that the file is auto-generated, and I'm not sure how to modify it.

When I try to run a program, the output is:
```bash
Arduino: 1.8.13 (Linux), Board: "Arduino Uno WiFi Rev2, ATMEGA328"

Sketch uses 1970 bytes (4%) of program storage space. Maximum is 48640 bytes.
Global variables use 28 bytes (0%) of dynamic memory, leaving 6116 bytes for local variables. Maximum is 6144 bytes.
An error occurred while uploading the sketch
avrdude: usbdev_open(): cannot open device: Permission denied
avrdude: jtag3_open_common(): Did not find any device matching VID 0x03eb and PID list: 0x2145
```

If this is an issue with the `udev` rules, how can I help to debug it?

PD: In the past I used this board in the same laptop. It was with an instance of Ubuntu 18.04. Back then I installed the Arduino IDE from the official package at the Arduino site. I remember that I also had issues, but I did not find the notes I took to fix them.

galgalesh, 2020-10-25:
Thanks for this detailed issue, @mmartinortiz!

The permission "Access USB hardware directly" gives Arduino access to the board _if your user account already has access_. I think the issue is that your user account does not yet have access.

I do not have this board so I need your help to test a fix. I wrote this based on the [comment you linked](https://forum.arduino.cc/index.php?topic=600558.msg4158135#msg4158135)

Run these two commands in a regular terminal as your user.

```bash
echo "SUBSYSTEM==\"usb\", MODE=\"0660\", GROUP=\"$(id -gn)\"" | sudo tee /etc/udev/rules.d/00-usb-permissions.rules
udevadm control --reload-rules
```

After you run this, try to run the program again.

mmartinortiz, 2020-10-25:
That did the trick :-)

Is this an issue related to my user (local installation) or my board? And could we modify the snap to add this rule?

In any case, thank you very much.

galgalesh, 2020-10-26:
> Is this an issue related to my user (local installation) or my board? 

This is an issue related to the board and the Arduino plugin for the board. Most plugins access boards using a standardized usb-serial protocol. By adding your user to the `dailout` group, your give it permission to use this protocol. That is part of the regular installation instructions.

The plugin of your board, however, accesses the USB interface _directly_. It does not use this standardized protocol. This direct access is normally only allowed by the `root` user because it is potentially dangerous and insecure. With the lines I pasted, you give your user unrestricted access to all raw USB interfaces.

> And could we modify the snap to add this rule?

Snaps are not permitted to add these rules because these rules change the security of your operating system. Users always have to add these rules themselves. Even if the snap could do it, I would not add it because it makes your system less secure and it is only needed for a very small number of boards.

What I will do is to add this information to the description of the snap, so that users who have this issue can more easily find the solution.

mmartinortiz, 2020-10-27:
Thanks for such a good explanation. Now it is clear to me, I created a pull request for adding this information to the Snap description.

PR #11 (2020-10-29): Updated information for specific boards
Based on your wonderful feedback for how to access the board directly, I wrote a small note on the Snap description for making this clear to other users.

Closes #10
closed 2020-10-29 by popey