snapcrafters/arduino #12: Arduino IDE startup error

neoaggelos, 2020-11-12:
### Summary

Arduino IDE will not startup on Ubuntu 20.04 with the following error:

```
$ arduino
Picked up JAVA_TOOL_OPTIONS: 
Set log4j store directory /home/neo/snap/arduino/47/.arduino15
java.lang.InternalError: java.lang.reflect.InvocationTargetException
	at java.desktop/sun.font.FontManagerFactory$1.run(FontManagerFactory.java:86)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.desktop/sun.font.FontManagerFactory.getInstance(FontManagerFactory.java:74)
	at java.desktop/java.awt.Font.getFont2D(Font.java:497)
	at java.desktop/java.awt.Font$FontAccessImpl.getFont2D(Font.java:237)
	at java.desktop/sun.font.FontUtilities.getFont2D(FontUtilities.java:158)
	at java.desktop/sun.font.FontUtilities.getCompositeFontUIResource(FontUtilities.java:395)
	at java.desktop/com.sun.java.swing.plaf.gtk.PangoFonts.lookupFont(PangoFonts.java:204)
	at java.desktop/com.sun.java.swing.plaf.gtk.GTKLookAndFeel.getGTKStyleFactory(GTKLookAndFeel.java:1582)
	at java.desktop/com.sun.java.swing.plaf.gtk.GTKLookAndFeel.loadStyles(GTKLookAndFeel.java:1546)
	at java.desktop/com.sun.java.swing.plaf.gtk.GTKLookAndFeel.initialize(GTKLookAndFeel.java:1439)
	at java.desktop/javax.swing.UIManager.setLookAndFeel(UIManager.java:586)
	at java.desktop/javax.swing.UIManager.setLookAndFeel(UIManager.java:633)
	at processing.app.Platform.setLookAndFeel(Platform.java:69)
	at processing.app.linux.Platform.setLookAndFeel(Platform.java:44)
	at processing.app.Base.<init>(Base.java:255)
	at processing.app.Base.main(Base.java:150)
Caused by: java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:490)
	at java.desktop/sun.font.FontManagerFactory$1.run(FontManagerFactory.java:84)
	... 16 more
Caused by: java.lang.NullPointerException
	at java.desktop/sun.awt.FontConfiguration.getVersion(FontConfiguration.java:1262)
	at java.desktop/sun.awt.FontConfiguration.readFontConfigFile(FontConfiguration.java:225)
	at java.desktop/sun.awt.FontConfiguration.init(FontConfiguration.java:107)
	at java.desktop/sun.awt.X11FontManager.createFontConfiguration(X11FontManager.java:719)
	at java.desktop/sun.font.SunFontManager$2.run(SunFontManager.java:379)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.desktop/sun.font.SunFontManager.<init>(SunFontManager.java:324)
	at java.desktop/sun.awt.FcFontManager.<init>(FcFontManager.java:35)
	at java.desktop/sun.awt.X11FontManager.<init>(X11FontManager.java:56)
	... 21 more
```

I get the error with revisions 41 (current stable) and 47 (current edge). It used to work with revision 32.

After googling the error a bit, it seems to be something fontconfig related, but I cannot try and fix it myself for the snap. Let me know if I can provide any more details.

### Specs

```
$ snap --version
snap    2.47.1
snapd   2.47.1
series  16
ubuntu  20.04
kernel  5.4.0-52-generic
```

```
$ snap info arduino --color=always | grep "installed:"
installed:          1.8.13            (47) 149MB -```
```

```
$ arduino --version
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Arduino: 1.8.13
```

galgalesh, 2020-11-12:
Can you show me the links that say this is fontconfig related?

galgalesh, 2020-11-12:
Can you revert to 32 to see if it still works with that version? If it's fontconfig related, it might be because of a change on your system instead of a change in the snap.

neoaggelos, 2020-11-21:
Hello,

Unfortunately this is not possible, since that snap revision has been removed from my machine.

However, I was able to download and run the Arduino IDE directly from https://www.arduino.cc/en/software. Installing the snap on a fresh machine worked as well, so this is probably a change on my system.

Thank you, closing the issue since it is probably not related to the snap itself.

closed 2020-11-21 by neoaggelos