snapcrafters/arduino #13: libusb 0.1.4 is missing in snap package

miamibc, 2020-11-17:
Error, while i'm trying to upload code to Digispark board:
```
micronucleus: error while loading shared libraries: libusb-0.1.so.4: cannot open shared object file: No such file or directory
```

Searching libusb* on my computer (Ubuntu 20.04) gave me these locations:
- /lib/i386-linux-gnu/libusb-0.1.so.4.4.4
- /lib/i386-linux-gnu/libusb-0.1.so.4
- /lib/x86_64-linux-gnu/libusb-0.1.so.4.4.4
- /lib/x86_64-linux-gnu/libusb-0.1.so.4
- /snap/core/10185/lib/x86_64-linux-gnu/libusb-0.1.so.4
- /snap/core/10185/lib/x86_64-linux-gnu/libusb-0.1.so.4.4.4
- /snap/core/10126/lib/x86_64-linux-gnu/libusb-0.1.so.4
- /snap/core/10126/lib/x86_64-linux-gnu/libusb-0.1.so.4.4.4

One of them must be used, but snap is blocking. Since it's not possible to link real directoryinto the snap filesystem, this must be fixed on your side.

Solved for me by installing not-a-snap version of Arduino IDE, and probably installing/reinstalling **libusb-0.1-4 libusb-dev** packages

galgalesh, 2020-11-18:
Thanks for the thorough bug report! Since I don't have this hardware, I'll need your help to test a fix. I released a version of Arduino which includes `libusb-0.1-4` to the edge channel.

You can install it by running `snap install arduino --edge`. To make sure you are using the snap version, you can run `snap run arduino` to test it out. Does this fix your issue?

miamibc, 2020-11-24:
Hello. 

No success yet.
```
Sketch uses 718 bytes (11%) of program storage space. Maximum is 6012 bytes.
Global variables use 9 bytes of dynamic memory.
Running Digispark Uploader...
Plug in device now... (will timeout in 60 seconds)
micronucleus: library/micronucleus_lib.c:66: micronucleus_connect: Assertion `res >= 4' failed.
Aborted (core dumped)
```

miamibc, 2020-11-24:
Works now. Thank you very much.

closed 2020-11-25 by miamibc