snapcrafters/arduino #15: Cany use ports on Arduino IDE

regus245, 2020-12-29:
Hi 
Could someone help me, I have installed the Arduino IDE on my laptop running Linux Ubuntu, I got a message  when I opened the IDE  station that I would not be able to load any boards as I will not have permissions to use the USB ports? 
I was told to run the following line in terminal " sudo usermod -a -G dialout <username>
When I run this line I get an error message saying that "dialout" does not exist.
dose anyone have a work round for this issue.

Many thanks

Regus

galgalesh, 2021-01-26:
Hi Regus. Could copy and paste the exact error message here? Thanks!

galgalesh, 2021-04-18:
Please reopen your issue if you can provide the exact info.

closed 2021-04-18 by galgalesh