snapcrafters/arduino #17: Unable to reach my arduino files in my USB drive (on Linux Ubuntu version 20

SkyGC, 2021-02-13:
hi.
I have an issue with the app.
When i want to open a project on my usb drive, Linux doesn't let me use the Arduino IDE instead of the notepad to open the file that I am looking for (see the first screenshot we can see that the Arduino IDE program doesn't appear (even if i scroll down the list))
[image]
[image]
[image]
[image]
(that's all the list that i have)

 and when I am using the Arduino IDE, I can't access to my usb drive to save the .ino file into it (see the second screenshot).
[image]
(as you can see, my usb drive doesn't appear)

Does someone have the same problem than me ?
Does someone knows how to fix it ?

(sorry if you can't understand my message. I'm french and i need a few english improvements )

galgalesh, 2021-02-15:
This is indeed an issue, but you can easily work around it. The USB drive should be in "Ordinateur" > "media" > "your username" > "Name of USB drive".

SkyGC, 2021-02-16:
Thanks.
This worked fine.
Now i can code without problems ^^!

galgalesh, 2021-02-17:
Reopening this since it is indeed a bug, even though we have a workaround

Monmoy042, 2021-07-10:
Is it okay to run "sudo usermod -a -G dialout kms" this command in my home directory??

galgalesh, 2021-07-19:
@Monmoy042 

I'm not entirely sure what you're asking. You can run this command wherever you like. It will add the user `kms` to the group `dailout`. This should give the user `kms` access to Arduino boards connected over USB tty.

Monmoy042, 2021-07-21:
@galgalesh yes it works perfectly in my system. I forgot to mention you that it worked in my system.

jnsgruk, 2026-11-11:
Should be resolved by #38