snapcrafters/arduino #18: Snap cannot access to USB

karpaz, 2021-04-14:
Hello, I have STM32F103C8, I'm flashing over ST-LINK V2  
When I flash over Arduino IDE (flatpak) it's alright, without any problem, but when I choose to use arduino (snap) than it's impossible to do that. I'd like to use snap, because it's seem to me better.

**Info**
Yes, I already used `sudo usermod -a -G dialout $USER` with restart, also `snap connections | grep arduino` is O.k with access to raw-usb, removable-media, unity7, wayland, x11 and more...

Yes, I already tried ``` echo "SUBSYSTEM==\"usb\", MODE=\"0660\", GROUP=\"$(id -gn)\"" | sudo tee /etc/udev/rules.d/00-usb-permissions.rules udevadm control --reload-rules ```

_So, I don't know what to do more..._

**Code in terminal (SNAP)**
```
snap run arduino
Picked up JAVA_TOOL_OPTIONS: 
Set log4j store directory /home/tasman/snap/arduino/50/.arduino15

(java:4298): Gtk-WARNING **: 23:17:01.781: Theme parsing error: gtk.css:1566:23: 'font-feature-settings' is not a valid property name

(java:4298): Gtk-WARNING **: 23:17:01.785: Theme parsing error: gtk.css:3616:25: 'font-feature-settings' is not a valid property name

(java:4298): Gtk-WARNING **: 23:17:01.786: Theme parsing error: gtk.css:4078:23: 'font-feature-settings' is not a valid property name

(java:4298): Gtk-WARNING **: 23:17:02.156: Theme parsing error: gtk.css:1566:23: 'font-feature-settings' is not a valid property name

(java:4298): Gtk-WARNING **: 23:17:02.160: Theme parsing error: gtk.css:3616:25: 'font-feature-settings' is not a valid property name

(java:4298): Gtk-WARNING **: 23:17:02.161: Theme parsing error: gtk.css:4078:23: 'font-feature-settings' is not a valid property name
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by processing.app.linux.GTKLookAndFeelFixer (file:/snap/arduino/50/lib/arduino-core.jar) to field com.sun.java.swing.plaf.gtk.GTKLookAndFeel.styleFactory
WARNING: Please consider reporting this to the maintainers of processing.app.linux.GTKLookAndFeelFixer
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Sketch uses 12780 bytes (39%) of program storage space. Maximum is 32768 bytes.
Global variables use 2064 bytes (20%) of dynamic memory, leaving 8176 bytes for local variables. Maximum is 10240 bytes.
/home/tasman/snap/arduino/50/.arduino15/packages/stm32duino/tools/stm32tools/2021.3.18/linux/stlink_upload: line 7: lsusb: command not found
USB Status [unknown]
/home/tasman/snap/arduino/50/.arduino15/packages/stm32duino/tools/stm32tools/2021.3.18/linux/stlink/st-flash: error while loading shared libraries: libusb-1.0.so.0: cannot open shared object file: No such file or directory
/home/tasman/snap/arduino/50/.arduino15/packages/stm32duino/tools/stm32tools/2021.3.18/linux/stlink_upload: line 35: lsusb: command not found
/home/tasman/snap/arduino/50/.arduino15/packages/stm32duino/tools/stm32tools/2021.3.18/linux/stlink_upload: line 7: lsusb: command not found
USB Status [unknown]
Waiting for tty device  

 should now be available.

```

**One screen is from flatpak and the other one with errors is from snap**
[flatpak](https://ibb.co/JrNLHnT)
[snap](https://ibb.co/0qwJhZf)

galgalesh, 2021-04-15:
Thanks for this very detailed bug report, very useful!

The problem seems to be that this package needs `libusb-1.0` while the snap has `libusb-0.1-4`. I'm looking for a solution and I'll let you know when I have more.

galgalesh, 2021-04-19:
@karpaz I pushed a version with a possible fix to the `edge` channel. Can you test it out? Thanks!

You can install it using `snap install --edge arduino` or `snap refresh --edge arduino`. Run `snap run arduino` during testing to make sure you're using the snap version and not the flatpak.

karpaz, 2021-04-19:
Yes, It's already fixed. Thanks a lot.

galgalesh, 2021-04-19:
This is now in stable

closed 2021-04-19 by galgalesh