snapcrafters/arduino #24: Won't install in Nvidia Jetpack 4.6

Ezward, 2021-12-04:
Attempting to install via the 'Software' application on Nvidia Jetpack 4.6 running on a Jetson Nano 2GB.
- The first time I tried this it did a lot of downloading and installation steps; perhaps 7 minutes or so.  It ended with the following popup message in the Software application 
```
Unable to install arduino:
cannot perform the following tasks:
```
- Now if I attempt to install I get that message after just about 20 seconds.
- Unfortunately there are no tasks listed.  Here is the syslog after the last boot and attempt to install:
```
Dec  4 12:35:05 diydonkeybot systemd[1]: Mounting Mount unit for arduino, revision 62...
Dec  4 12:35:05 diydonkeybot kernel: [   89.933517] squashfs: SQUASHFS error: Filesystem uses "lzo" compression. This is not supported
Dec  4 12:35:05 diydonkeybot mount[7375]: mount: /snap/arduino/62: wrong fs type, bad option, bad superblock on /dev/loop6, missing codepage or helper program, or other error.
Dec  4 12:35:05 diydonkeybot systemd[1]: snap-arduino-62.mount: Mount process exited, code=exited status=32
Dec  4 12:35:05 diydonkeybot systemd[1]: snap-arduino-62.mount: Failed with result 'exit-code'.
Dec  4 12:35:05 diydonkeybot systemd[1]: Failed to mount Mount unit for arduino, revision 62.
Dec  4 12:35:05 diydonkeybot systemd[1]: Reloading.
Dec  4 12:35:06 diydonkeybot systemd[1]: Starting resolvconf-pull-resolved.service...
Dec  4 12:35:06 diydonkeybot systemd[1]: Reloading.
Dec  4 12:35:06 diydonkeybot systemd[1]: Started resolvconf-pull-resolved.service.
Dec  4 12:35:06 diydonkeybot snapd[4247]: taskrunner.go:271: [change 4 "Mount snap \"arduino\" (62)" task] failed: systemctl command [start snap-arduino-62.mount] failed with exit status 1: Job for snap-arduino-62.mount failed.
Dec  4 12:35:06 diydonkeybot snapd[4247]: See "systemctl status snap-arduino-62.mount" and "journalctl -xe" for details.
Dec  4 12:35:08 diydonkeybot snapd[4247]: handlers.go:642: Reported install problem for "arduino" as aaf25394-5541-11ec-bd32-fa163e6cac46 OOPSID
Dec  4 12:35:43 diydonkeybot dbus-daemon[6306]: [session uid=1000 pid=6306] Activating service name='org.freedesktop.Notifications' requested by ':1.42' (uid=1000 pid=7509 comm="/usr/lib/chromium-browser/chromium-browser --enabl")
Dec  4 12:35:44 diydonkeybot dbus-daemon[6306]: [session uid=1000 pid=6306] Successfully activated service 'org.freedesktop.Notifications'
Dec  4 12:37:26 diydonkeybot snapd[4247]: api_snaps.go:307: Installing snap "arduino" revision unset
Dec  4 12:37:32 diydonkeybot systemd[1]: Reloading.
Dec  4 12:37:33 diydonkeybot systemd[1]: Starting resolvconf-pull-resolved.service...
Dec  4 12:37:33 diydonkeybot systemd[1]: Reloading.
Dec  4 12:37:33 diydonkeybot systemd[1]: Started resolvconf-pull-resolved.service.
Dec  4 12:37:33 diydonkeybot systemd[1]: Mounting Mount unit for arduino, revision 62...
Dec  4 12:37:33 diydonkeybot mount[8095]: mount: /snap/arduino/62: wrong fs type, bad option, bad superblock on /dev/loop6, missing codepage or helper program, or other error.
Dec  4 12:37:33 diydonkeybot systemd[1]: snap-arduino-62.mount: Mount process exited, code=exited status=32
Dec  4 12:37:33 diydonkeybot systemd[1]: snap-arduino-62.mount: Failed with result 'exit-code'.
Dec  4 12:37:33 diydonkeybot kernel: [  238.037740] squashfs: SQUASHFS error: Filesystem uses "lzo" compression. This is not supported
Dec  4 12:37:33 diydonkeybot systemd[1]: Failed to mount Mount unit for arduino, revision 62.
Dec  4 12:37:33 diydonkeybot systemd[1]: Reloading.
Dec  4 12:37:34 diydonkeybot systemd[1]: Starting resolvconf-pull-resolved.service...
Dec  4 12:37:34 diydonkeybot systemd[1]: Reloading.
Dec  4 12:37:34 diydonkeybot systemd[1]: Started resolvconf-pull-resolved.service.
Dec  4 12:37:34 diydonkeybot snapd[4247]: taskrunner.go:271: [change 5 "Mount snap \"arduino\" (62)" task] failed: systemctl command [start snap-arduino-62.mount] failed with exit status 1: Job for snap-arduino-62.mount failed.
Dec  4 12:37:34 diydonkeybot snapd[4247]: See "systemctl status snap-arduino-62.mount" and "journalctl -xe" for details.
Dec  4 12:37:34 diydonkeybot snapd[4247]: handlers.go:642: Reported install problem for "arduino" as already-reported
```
- Here is my  jetpack version:
```
$ sudo apt-cache show nvidia-jetpack
Package: nvidia-jetpack
Version: 4.6-b199
Architecture: arm64
Maintainer: NVIDIA Corporation
Installed-Size: 194
Depends: nvidia-cuda (= 4.6-b199), nvidia-opencv (= 4.6-b199), nvidia-cudnn8 (= 4.6-b199), nvidia-tensorrt (= 4.6-b199), nvidia-visionworks (= 4.6-b199), nvidia-container (= 4.6-b199), nvidia-vpi (= 4.6-b199), nvidia-l4t-jetson-multimedia-api (>> 32.6-0), nvidia-l4t-jetson-multimedia-api (<< 32.7-0)
Homepage: http://developer.nvidia.com/jetson
Priority: standard
Section: metapackages
Filename: pool/main/n/nvidia-jetpack/nvidia-jetpack_4.6-b199_arm64.deb
Size: 29368
SHA256: 69df11e22e2c8406fe281fe6fc27c7d40a13ed668e508a592a6785d40ea71669
SHA1: 5c678b8762acc54f85b4334f92d9bb084858907a
MD5sum: 1b96cd72f2a434e887f98912061d8cfb
Description: NVIDIA Jetpack Meta Package
Description-md5: ad1462289bdbc54909ae109d1d32c0a8

Package: nvidia-jetpack
Version: 4.6-b197
Architecture: arm64
Maintainer: NVIDIA Corporation
Installed-Size: 194
Depends: nvidia-cuda (= 4.6-b197), nvidia-opencv (= 4.6-b197), nvidia-cudnn8 (= 4.6-b197), nvidia-tensorrt (= 4.6-b197), nvidia-visionworks (= 4.6-b197), nvidia-container (= 4.6-b197), nvidia-vpi (= 4.6-b197), nvidia-l4t-jetson-multimedia-api (>> 32.6-0), nvidia-l4t-jetson-multimedia-api (<< 32.7-0)
Homepage: http://developer.nvidia.com/jetson
Priority: standard
Section: metapackages
Filename: pool/main/n/nvidia-jetpack/nvidia-jetpack_4.6-b197_arm64.deb
Size: 29356
SHA256: 104cd0c1efefe5865753ec9b0b148a534ffdcc9bae525637c7532b309ed44aa0
SHA1: 8cca8b9ebb21feafbbd20c2984bd9b329a202624
MD5sum: 463d4303429f163b97207827965e8fe0
Description: NVIDIA Jetpack Meta Package
Description-md5: ad1462289bdbc54909ae109d1d32c0a8
```

Ezward, 2021-12-04:
I was able to install the Arduino IDE using these instructions: https://www.jetsonhacks.com/2019/10/04/install-arduino-ide-on-jetson-dev-kit/

diddledani, 2021-12-04:
Your operating system doesn't support the required kernel features to run this, and many other, snap(s). The key is this line in your log:

```
[   89.933517] squashfs: SQUASHFS error: Filesystem uses "lzo" compression. This is not supported
```

You need to contact the provider of your operating system to get the feature enabled or find a different operating system.

galgalesh, 2021-12-19:
Closing this as it's not a snap issue, but an issue of your device not supporting `lzo` compression, which is needed for many/most desktop snaps.

closed 2021-12-19 by galgalesh