snapcrafters/arduino #26: [Bug]: Can't find adafruit-nrfutil on path

grantwest, 2022-01-19:
### What happened?

I'm using these board extensions: https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
With the Adafruit nRF52 extension installed.
Target board: Nordic nRF52840 DK

Run this command to install a required programming utility: 
```
arduino.pip install adafruit-nrfutil
```

When I click the "Verify" button I get this error:
```
Arduino: 1.8.15 (Linux), Board: "Nordic nRF52840 DK, S140 6.1.1, Level 0 (Release), Serial"

exec: "adafruit-nrfutil": executable file not found in $PATH
Error compiling for board Nordic nRF52840 DK.
```

### What should have happened?

The app should be able to find the `adafruit-nrfutil` in the path. 

### Output of `snap info $snap_name`

```shell
name:      arduino
summary:   Write code and upload it to your Arduino-compatible board.
publisher: Merlijn Sebrechts
store-url: https://snapcraft.io/arduino
contact:   https://github.com/snapcrafters/arduino/issues
license:   GPL-2.0
description: |
  Arduino is an open-source physical computing platform based on a simple I/O board and a
  development environment that implements the Processing/Wiring language. Arduino can be used to
  develop stand-alone interactive objects or can be connected to software on your computer (e.g.
  Flash, Processing and MaxMSP). The boards can be assembled by hand or purchased preassembled at
  https://arduino.cc
  
  **Setup**
  
  In order to upload code an arduino board over USB, you need to add your user to the `dialout`
  group. Open a terminal window, run the following command and *reboot* your computer.
  
  > `sudo usermod -a -G dialout $USER`
  
  Now restart your computer and you're good to go!
  
  If you have one of the following boards, you need to grant specific permissions to your user for
  accessing directly the USB. This issue is more detailed
  [here](https://github.com/snapcrafters/arduino/issues/10#issuecomment-716683018)
  
  - Arduino Uno WiFi Rev2, ATMEGA328
  
  For granting this direct access:
  
  
  echo "SUBSYSTEM==\"usb\", MODE=\"0660\", GROUP=\"$(id -gn)\"" | sudo tee
  /etc/udev/rules.d/00-usb-permissions.rules
  udevadm control --reload-rules
  ```
  
  **Python 3 support**
  
  This version of Arduino IDE runs in a sandbox for improved security and stability. As a result, it
  does not have access to the Python libraries on your device.
  
  Python 3 and the `serial` library are installed inside of the sandbox so they can be used by
  plugins. If a plugin needs additional Python 3 libraries, you can install them using the
  `arduino.pip` command. For example, the following command installs the `requests` library inside
  of the sandbox so the Arduino IDE can use it.
  
  > `arduino.pip install requests`
  
  Python 2 is currently not supported in this package because it reached end of life on January 1st
  2020. If you have hardware that _requires_ Python 2 support, please let us know.
  
  **Are you having issues?**
  
  Let us know by creating a new issue here: https://github.com/snapcrafters/arduino/issues
commands:
  - arduino
  - arduino.pip
snap-id:      q7ex9QXSEs0gfmZHBPYtmSGSwjDTuYhF
tracking:     latest/stable
refresh-date: today at 15:55 CST
channels:
  latest/stable:    1.8.15 2021-07-08 (61) 207MB -
  latest/candidate: ↑                            
  latest/beta:      ↑                            
  latest/edge:      1.8.19 2021-12-29 (70) 204MB -
installed:          1.8.15            (61) 207MB -
```

### Output of `snap connections $snap_name`

```shell
Interface                 Plug                     Slot                             Notes
content[gnome-3-28-1804]  arduino:gnome-3-28-1804  gnome-3-28-1804:gnome-3-28-1804  -
content[gtk-3-themes]     arduino:gtk-3-themes     gtk-common-themes:gtk-3-themes   -
content[icon-themes]      arduino:icon-themes      gtk-common-themes:icon-themes    -
content[sound-themes]     arduino:sound-themes     gtk-common-themes:sound-themes   -
desktop                   arduino:desktop          :desktop                         -
desktop-legacy            arduino:desktop-legacy   :desktop-legacy                  -
gsettings                 arduino:gsettings        :gsettings                       -
home                      arduino:home             :home                            -
network                   arduino:network          :network                         -
network-bind              arduino:network-bind     :network-bind                    -
opengl                    arduino:opengl           :opengl                          -
raw-usb                   arduino:raw-usb          :raw-usb                         -
removable-media           arduino:removable-media  :removable-media                 -
serial-port               arduino:serial-port      -                                -
unity7                    arduino:unity7           :unity7                          -
wayland                   arduino:wayland          :wayland                         -
x11                       arduino:x11              :x11                             -
```

### Output of `snap version`

```shell
snap    2.53+21.10ubuntu1
snapd   2.53+21.10ubuntu1
series  16
ubuntu  21.10
kernel  5.13.0-25-generic
```

grantwest, 2022-01-19:
It looks like `.local/bin` needs to be added to the path, but I don't know enough about snap to try it out. Is there a way I can append to the path with snap?

AD-Wright, 2022-06-16:
Also having this issue... is it as simple as adding to this line?: https://github.com/snapcrafters/arduino/blob/645927d8277a548d8a008d7d100d49f240e73b09/scripts/arduino-snap#L31  

I also am not familiar at all with snap.

closed 2022-11-08 by galgalesh