snapcrafters/arduino #32: [Enhancement]: Arduino IDE 2.0.x support

d-beamon, 2022-12-09:
### What changes would you like?

There is a new version of the Arduino IDE (v2.0.x). This version includes many improvements like Arduino Cloud integration and an improved UX.

The old Arduino v1.8.x will be eventually discontinued and no longer supported, so it would be great to have the new v2.0.x version snapped.

merlijn-sebrechts, 2022-12-09:
This would probably be a separate snaps. That's also how the flatpak works. I don't have time to work on this currently, but any contributions are welcome!

jnsgruk, 2026-11-11:
Merged #37 - I'll be opening a new `2` track soon so you'll be able to `snap install arduino --channel 2/stable`.

Once it's been tested I'm minded to make the `2` track the default.

closed 2026-11-11 by jnsgruk