snapcrafters/arduino #33: [Bug]: Python 3.6 problem with some library's

z37a, 2023-06-25:
### What happened?

When you add the latest 8266 support from the IDE, you cant compile anything for these family boards. That's happens because the snap use Python 3.6, and these library's require 3.7 or newest.

### What should have happened?

When you upload a new sketch using 8266 library

### Output of `snap info $snap_name`

```shell
name:      arduino
summary:   Write code and upload it to your Arduino-compatible board.
publisher: Snapcrafters✪
store-url: https://snapcraft.io/arduino
contact:   https://github.com/snapcrafters/arduino/issues
license:   GPL-2.0
description: |
  Arduino is an open-source physical computing platform based on a simple I/O
  board and a development environment that implements the Processing/Wiring
  language. Arduino can be used to develop stand-alone interactive objects or
  can be connected to software on your computer (e.g. Flash, Processing and
  MaxMSP). The boards can be assembled by hand or purchased preassembled at
  https://arduino.cc
  
  **Setup**
  
  In order to upload code an arduino board over USB, you need to add your
  user to the `dialout` group. Open a terminal window, run the following
  command and *reboot* your computer.
  
  > `sudo usermod -a -G dialout $USER`
  
  Now restart your computer and you're good to go!
  
  If you have one of the following boards, you need to grant specific
  permissions to your user for accessing directly the USB. This issue is more
  detailed
  [here](https://github.com/snapcrafters/arduino/issues/10#issuecomment-71668
  3018)
  
  - Arduino Uno WiFi Rev2, ATMEGA328
  
  For granting this direct access:
  
  
  echo "SUBSYSTEM==\"usb\", MODE=\"0660\", GROUP=\"$(id -gn)\"" | sudo tee
  /etc/udev/rules.d/00-usb-permissions.rules
  udevadm control --reload-rules
  ```
  
  **Python 3 support**
  
  This version of Arduino IDE runs in a sandbox for improved security and
  stability. As a result, it does not have access to the Python libraries on
  your device.
  
  Python 3 and the `serial` library are installed inside of the sandbox so
  they can be used by plugins. If a plugin needs additional Python 3
  libraries, you can install them using the `arduino.pip` command. For
  example, the following command installs the `requests` library inside of
  the sandbox so the Arduino IDE can use it.
  
  > `arduino.pip install requests`
  
  Python 2 is currently not supported in this package because it reached end
  of life on January 1st 2020. If you have hardware that _requires_ Python 2
  support, please let us know.
  
  **Are you having issues?**
  
  Let us know by creating a new issue here:
  https://github.com/snapcrafters/arduino/issues
commands:
  - arduino
  - arduino.pip
snap-id:      q7ex9QXSEs0gfmZHBPYtmSGSwjDTuYhF
tracking:     latest/stable
refresh-date: ayer a las 23:00 -03
channels:
  latest/stable:    1.8.19 2022-12-04 (85) 205MB -
  latest/candidate: ↑                            
  latest/beta:      1.8.19 2022-12-04 (85) 205MB -
  latest/edge:      1.8.19 2022-12-08 (91) 205MB -
installed:          1.8.19            (85) 205MB -
```

### Output of `snap connections $snap_name`

```shell
Interfaz                  Enchufe                  Ranura                           Notas
content[gnome-3-28-1804]  arduino:gnome-3-28-1804  gnome-3-28-1804:gnome-3-28-1804  -
content[gtk-3-themes]     arduino:gtk-3-themes     gtk-common-themes:gtk-3-themes   -
content[icon-themes]      arduino:icon-themes      gtk-common-themes:icon-themes    -
content[sound-themes]     arduino:sound-themes     gtk-common-themes:sound-themes   -
desktop                   arduino:desktop          :desktop                         -
desktop-legacy            arduino:desktop-legacy   :desktop-legacy                  -
gsettings                 arduino:gsettings        :gsettings                       -
home                      arduino:home             :home                            -
network                   arduino:network          :network                         -
network-bind              arduino:network-bind     :network-bind                    -
opengl                    arduino:opengl           :opengl                          -
raw-usb                   arduino:raw-usb          :raw-usb                         -
removable-media           arduino:removable-media  :removable-media                 -
serial-port               arduino:serial-port      -                                -
unity7                    arduino:unity7           :unity7                          -
wayland                   arduino:wayland          :wayland                         -
x11                       arduino:x11              :x11                             -
```

### Output of `snap version`

```shell
snap    2.59.5
snapd   2.59.5
series  16
ubuntu  22.04
kernel  5.19.0-45-generic
```

### Relevant log output

```shell
Arduino:1.8.19 (Linux), Tarjeta:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

/home/z37a/snap/arduino/85/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/tools/mkbuildoptglobals.py
Minimal supported version of Python is 3.7
exit status 1
Error compilando para la tarjeta NodeMCU 1.0 (ESP-12E Module).
```

### Teminal output of app

```shell
arduino.pip --version
pip 21.3.1 from /snap/arduino/85/lib/python3.6/site-packages/pip (python 3.6)
```

amici, 2023-07-01:
I have the same issue, with `pip 21.3.1 from /snap/arduino/85/lib/python3.6/site-packages/pip (python 3.6)`, Ubuntu v20.04

edysli, 2023-07-30:
I also hit this issue as the [Arduino core for ESP8266 chips](https://github.com/esp8266/Arduino) now require Python 3.7 or newer.

dobicinaitis, 2023-12-23:
I just encountered the same issue. While the bug is not fixed, as a workaround, I managed to get things working by switching to the older `3.0.2` version of the [Arduino core for ESP8266 WiFi chip](https://github.com/esp8266/Arduino) library.

closed 2026-11-11 by jnsgruk